ukmars/ukmarsbot #1: Initial Test Issue to verify function

IanSuper7, 2019-07-13:
An examination of issues to see whether they are suitable for managing consultation feedback.

IanSuper7, 2019-07-13:
What do you think Peter

micromouseonline, 2019-07-13:
This is the way most github projects manage things. It is like a forum in that the history is all there and there are not ever-growing email lists with members forgetting to 'reply all'.

Email discussions are also useful though because everything on here is public.

IanSuper7, 2019-07-13:
Looked into this to see if it was the best way to invite input in the collaboration periods. I think it is because once an issue has been raised and discussed it will be visible for all and could reduce a lot of overlapping and potentially conflicting emails. Agree there will be other discussions best conducted by email. My intention is to write this into the contributors guide as the vehicle for suggesting improvements in the collaboration windows.

closed 2019-07-16 by IanSuper7