ukmars/ukmarsbot #2: Bill of Materials out of Date and files out of step

IanSuper7, 2019-07-15:
**Project Phase**  Revision 1 Requirements
**Artifacts affected -** BOM.md and BOM excel spreadsheet
**Issue:**
BOM excel spreadsheet is out of date for motor driver choice.
BOM.md is out of sync with the excel spreadsheet
**Suggestion:**
Update excel spreadsheet for motor driver and check other items are still valid on the spreadsheet.
Consider demising .md file as it only adds to maintenance and most people will be able to deal with an.xlsx file

micromouseonline, 2019-07-15:
Agree. Only one BOM file should be present and that might as well be the Excel spreadsheet.

I will delete the .md file.

Who is to be the maintainer of the BOM?

davidhannaford, 2019-07-16:
I like the idea of using the excel file as the master

IanSuper7, 2019-07-16:
Ok so I've updated the BOM for the motor driver and dded a battery clip. Ongoing maintenance in the short term I think should be with the person drawing up the Eagle files, so Peter for the mother board and the maze sensor board and me for the line sensor board.

ceeaspb, 2019-09-15:
The 74ls86n is not included. I do think it would be useful to add this as I would like to know where people sourced it from. Ebay has various sellers but it's not clear which if any are ok. I couldn't find it on Farnell or Rapid.

Also whether you soldered it directly to be board or likely used an IC socket.

micromouseonline, 2019-09-15:
Yes, it is board-only at this stage. I should probably have included one when I sent it out.

Note also that I think we have the wrong Diode specified. Almost any power diode that will fit should do. The 1N5407 will be too big though.

It needs to be soldered to the board as there will not likely be room if it is socketed.

Farnell have them:

http://uk.farnell.com/jsp/search/productdetail.jsp?CMP=i-ddd7-00001003&sku=1740036

You can get them reliably from eBay though.

These are possible sellers (I know nothing about them except that they seem to be UK):

https://www.ebay.co.uk/itm/5-x-DM74LS86N-74LS86-DIP14-ORIGINAL-NSC-PARTS-UK-STOCK/223651549578?hash=item3412abd18a:g:GhoAAOSw2mddcNkH

https://www.ebay.co.uk/itm/5-x-SN74LS86N-SN74LS86AN-74LS86-Motorola-Texas-Instruments-5-Per-Sale-1ea/222114887240?hash=item33b7143648:g:BOsAAOSw241Yf0wx

https://www.ebay.co.uk/itm/74LS86N-SIGNETICS-INTEGRATED-CIRCUIT-X5pcs/261169894940?epid=1928189317&hash=item3ccef0561c:g:qdAAAMXQKQhRHOBt

—
Peter Harrison
peter.harrison@helicron.net

closed 2022-08-16 by micromouseonline