ukmars/ukmarsbot #3: Is the Arduino Capable of Supporting a Maze Solver

IanSuper7, 2019-07-18:
**This issue has been raised by Ian to publish a query raised by Duncan and subsequent comments from others. All further comments on this thread should be added directly to this issue.** This Issue uses the preferred format for raising issues on UKMARSBOT

**Project Phase** Revision 1 Requirements
**Artifacts affected -** requirements.md and BOM excel spreadsheet
**Issue:**
My only concern is that the ATMEGA used in the Arduino Nano has only 2K of Ram. While it should be possible to make a solver under this constraint it does seem unnecessarily restrictive for a first effort at a solver.
**Suggestion:**
Review the choice of the Arduino as the microcontroller.

IanSuper7, 2019-07-18:
**David T added the following comment:**
In theory the memory is enough for the standard maze, but if you want to progress to the half size maze then there are bigger problems especially with flood fill values going the 255 single byte boundary. 
I am invertigating connecting some FRAM via the SPI connection. I bought some 8k modules to add to my mega328 chips. I've not looked at the design yet, but exposing the SPI lines for add-ons might be an idea.

IanSuper7, 2019-07-18:
**Derek added the following comments:**
This mouse is not small enough for a ½ size mouse so its not a problem being unable to store a map for a 1/2 size maze.
The Arduino uses standard C ish so experience on this should be ok for bigger things, it has 1K of eeprom you can put the maze map in directly saving some space.

The I2c and SPI should be left clear if possible, this may not be possible with SPI as extra pins are required for chip select, I2c can be used to add a display or extra sensors easily

micromouseonline, 2019-07-18:
There should be enough room in the nano for a maze solver.  A wall map needs only 64 bytes and a simple distance flood takes 256 bytes.

There is also the new Arduino Nano Every processor which should work in the same socket and gives 6k RAM.

https://store.arduino.cc/nano-every

Also able to sit in the same socket are the STM32 Nucleo 32 boards which have a huge amount of power compared to the Arduino nano.

The basic design of the machine has been kept small enough to navigate through the maze without too much difficulty. It will be narrow enough, for example, to pass along diagonal paths if desired.

micromouseonline, 2019-07-18:
I2C is available on the sensor connector and can be used to substantially expand the IO capabilities to include more digital ports, displays and even a gyro and accelerometer.

micromouseonline, 2022-08-16:
The Mazerunner software demonstrates a full solver.
https://github.com/ukmars/ukmarsbot-mazerunner

closed 2022-08-16 by micromouseonline