ukmars/ukmarsbot #10: Add 'copper pours' as ground planes top and bottom to sensor PCB

gbulmer, 2019-08-02:
Urgency: Revision 1.

Impacted artefact: line_sensor_1p1_brd.pdf

There may be copper pours as ground planes already on the PCB, but I can't see the evidence in the line_sensor_1p1_brd.pdf file. If they are there, but omitted to make the traces clear, add a second rendition showing the copper pours.

Copper pours top and bottom will reduce ambient light, which should improve the consistency and performance of the line sensor.

Using those copper pours as ground planes will likely improve ground connections, and reduce sensitivity to EMI.

micromouseonline, 2019-08-02:
On the main board, there is intentionally no copper pour on the top surface. In terms of EMC, it seemed redundant in this application and I did not want there to be any copper under the motors where vibration may wear away the solder mask. Probably not a problem though.

At the front of the main board, the copper stops at the point where a builder might choose to cut off part of the board if they wanted the space for anything else. Sumo builders may want that. I think we have assumed that sensor boards are responsible for their own issues.

IanSuper7, 2019-08-02:
Deliberately omitted copper pours to make the traces clearer for initial comment. Board has copper pours so will post these renditions also.

closed 2019-08-02 by IanSuper7

gbulmer, 2019-08-02:
The issue is raised for the Line Sensor board, and not ukmarsBot (yet :-). 
I don't want to cause any extra delay to ukmarsBot Revision 1 PCB if I can avoid it.

> On the main board, there is intentionally no copper pour on the top surface. 

I appreciate it might be undesirable to have a top surface copper pour **_under the motors_** which might eventually rub away the soldermask.
> In terms of EMC, it seemed redundant in this application

Anything which might help reduce emission of EM radiation seems helpful, especially if its very cheap. 
> and I did not want there to be any copper under the motors where vibration may wear away the solder mask. 

It's what I had assumed. However a thin tape would solve it if it becomes an issue.

> Probably not a problem though.

I think the motors are quite smooth on the outside, so they are unlikely to cause an issue. The gearboxes might though.
> 
> At the front of the main board, the copper stops at the point where a builder might choose to cut off part of the board if they wanted the space for anything else. Sumo builders may want that. 

That makes sense. I accept that.

> I think we have assumed that sensor boards are responsible for their own issues.

Agreed. Hence this issue is only raised for the Line Sensor PCB.

gbulmer, 2019-08-02:
While you clearly have the functionality to close the issue, there several
ways to use the issue tracking system that I know about.

1. Allow the person/entity who raised the issue to close the issue. This
acts as a sort of public 'protocol handshake' to ensure both parties agree,
and everyone can see that agreement has been reached.

2. Declare when the issue will be resolved (eg at a milestone) and leave
the issue open.
Then when the milestone is reached, the issues which have been resolved are
closed. The open issue acts as a public acknowledgment of the issue (making
it easy for people to see that the issue is already identified and accepted
potentially saving duplication), as timing for resolution (which is very
handy), and effectively a 'Feature request' which will consume resources
and hence may cause rescheduling.

3. Close the issue when it's publicly resolved, ie the of the issue
resolution is published. IMHO this seems clear and unambiguous, and is a
simplified approach to case 2.

If there is any lack of clarity about the process being used, it might be
worth adding some notes about the process to the contributors guidelines at
some stage.