ukmars/ukmarsbot #11: Add "Part Names" to sensor PCB

gbulmer, 2019-08-02:
Urgency: Revision 1
Impacted artefacts: line_sensor_1p1_brd.pdf and its underlying PCB ECAD.

Every part **_must_** have the 'Part Name' or 'Part ID' as its legend on the silkscreen to identify it, and unambiguously cross reference it to the schematic, BoM and Build Guide, Testing & Debugging documentation, etc.

Without this, it is harder to cross-reference. Worse, it is hard to describe bugs or ask question purely with text. Text is straightforward to search, helping anyone discover if a bug related to one or more parts is known, or there is further information on that part.

The part legends on the PCB appear to be inconsistent. In some cases there is only part value, in others only part name, and in others both.

Having the part value as well as the Part Name is helpful. I fully support that as an aim. 
However Part Name is necessary and sufficient.

micromouseonline, 2019-08-07:
I believe this is now done.

gbulmer, 2019-08-07:
Agree. I'll close.

closed 2019-08-07 by gbulmer