ukmars/ukmarsbot #12: Put legends for connections and information on both sides of sensor PCB

gbulmer, 2019-08-02:
Urgency: Revision 1
Impacted artefacts: line_sensor_1p1_brd.pdf and its underlying PCB ECAD

I am assuming the render is from the component side, and hence the surface which will be facing downwards when attached to the ukmarsBot. If I am making the wrong assumption, then there is a different but closely related set of issues.

### Assuming the sensor PCB render is the components side

When attached to the ukmars bot, the user will need to connect the ukmarsBot signals to the sensor board. However, there is no silkscreen on the visible side of the board to help. 

While the header pins might align perfectly, it is still helpful to help the user recognise that they are connecting it correctly. So add designators/legends to the visible side of the sensor PCB.

Also, the UKMARS name and URL should be visible on the top surface to help 'build the brand' and help people find out more.

IanSuper7, 2019-08-03:
Most components will be facing upwards with the silk screen visible, this includes the connector to the mother board. The 3 emitters and 4 phototransistors are on the opposite face of the board (bottom) soldered through to the top  so that they are "looking down" onto the line and the markers.

gbulmer, 2019-08-03:
Okay. I understand.

In that case all of the components except 'the 3 emitters and 4 phototransistors' are on the wrong side of the PCB. They need to be mirrored. Then the silkscreen outlines, part names part values and labels will be visible on the side where the parts are fitted and text will be printed the correct way round (it will look mirrored from the front).

IanSuper7, 2019-08-04:
v1.2 line sensor board layout will address parts on both sides of the board

gbulmer, 2019-08-04:
It is **_much_** clearer which components are mounted on which side. Excellent.

It is worth having a third rendering of the PCB showing it without the copper pours having been filled in (with the copper pour polygon outlines). That way it is easy to see that the components mounted on the bottom have their silk screen legends mirrored.

The files renders/line_sensor_1p1_brd.pdf and renders/line_sensor_1p1_sch.pdf should be removed as they are inconsistent with 1p2, and hence incorrect.

micromouseonline, 2019-08-07:
The renders are now consistent and the board views are derived from the gerbers rather than the Eagle .brd files so they are more representative of what the finished boards will look like.

gbulmer, 2019-08-07:
They are much clearer. Good work everyone.

It's hard to be sure, as the boards look like they are scaled up. However, the silkscreen text height might be a bit small on the Line sensor. However, that's a separate issue.
 
I will close this issue.

closed 2019-08-07 by gbulmer