ukmars/ukmarsbot #15: Outline of Line Sensor PCB might be 'un-manufacturable'

gbulmer, 2019-08-02:
Urgency: Revision 1
Artefacts impacted: line_sensor_1p1_brd.pdf and its underlaying PCB ECAD

The outline of the Line Sensor PCB has two sharp, 90º concave corners.
I believe that is not possible to manufacture.

It can't be made using a V-Cut as that is from edge to edge and so would cut across parts of the PCB, severing tracks, and weakening the PCB.

A manufacturer will specify the size of router they use to cut around PCBs. It is common to use a 2mm diameter routers. Therefore those corners should have a radius of 1mm.

IanSuper7, 2019-08-02:
Will modify design to include rounded concave corners

closed 2019-08-02 by IanSuper7

micromouseonline, 2019-08-02:
It has been my experience that board houses have no problem with that kid of outline. They simply do the best they can with the cutter they specify. The resulting board is still fine for our purposes.

The mainboard has the same issue and I am unconcerned. All my mouse boards have this same thing.

BTW - internal cutouts will be done the same way.

—
Peter Harrison
peter.harrison@helicron.net

gbulmer, 2019-08-03:
Okay. Given this is a prototype for small quantity, I can accept that.

However, I think it's better to ensure all PCBs are 'as specified', and not
subject to "they simply do the best they can with the cutter they specify."
Further, there is an approach which recommends trying to get what is
manufactured close to the final goal just so that there are fewer surprises
late on.