ukmars/ukmarsbot #5: Add Open Source Hardware (OSH) Logo to PCB

gbulmer, 2019-08-02:
IMHO not urgent. Revision 2 or later.

Clearly mark the UKMaRSBot PCB so show that it is Open Hardware. Hence the PCB CAD is impacted. I have seen projects where the logo is added to the schematic too

A simple way to do that is put the [Open Hardware logo](https://www.oshwa.org/open-source-hardware-logo/) on the PCB silkscreen. If practical, I'd put it into the copper too.

The logo is available from [OSHWA.org in various formats](https://www.oshwa.org/open-source-hardware-logo/)

An [Eagle ECAD 'ULP'](https://github.com/WestfW/Eagle-lbr-published/blob/master/oshw-logo.ulp) for the  logo.
An [article](http://darkerview.com/wordpress/?p=20495) which explains and links to an [Eagle library part](http://darkerview.com/wordpress/wp-content/uploads/2016/11/oshw.zip)

There are versions for other ECAD tools.

The logo is CC-SA, so I am certain there is no associated cost. 

However I have not done _rigorous_ research to ensure the spirit and [definition of OSH](https://www.oshwa.org/definition/) is matched. Though I struggle to imagine anyway UKMaRSBot wouldn't be compatible.

micromouseonline, 2019-08-02:
At present, based on a quick look at the definition,  we are not very compliant with the OSHW recommendations. Something to look at for the future.

gbulmer, 2019-08-02:
I think carrying the logo would encourage/enable a bunch of web sites to carry articles about the 'bot, and hence reach more people.

I'll leave the issue open for now.