ukmars/ukmarsbot #7: Add more specific URL to PCB

gbulmer, 2019-08-02:
Not urgent. Revision 2 would be okay.

Adding a more direct URL to make straightforward for anyone seeing a UKMaRSBot, even a picture of it, to follow up, and find out more.

The URL should either be to its repo at github/ukmars/ukmarsbot, or a specific landing page at ukmars.org,  eg ukmars.org/ukmarsbot.

This would impact the PCB CAD, and if it is not a Github URL, require some work on the ukmars.org web site.

micromouseonline, 2019-08-02:
So - what do we think?

In time, all the useful information about the 'bot should probably be in the repo's wiki. Perhaps that would be a good landing page.

The URL needs to be short really. I am keen that even relatively poor photos will have a legible URL to guide people to further information. Very often one sees pictures of interesting things with no clear way to find out more.

Both locations will be covered up in a race-ready 'bot. I will probably place another underneath long with the society logo. our own sensor boards should probably carry a replacement for the name and link if they cover up the one on the main board.

gbulmer, 2019-08-02:
> So - what do we think?
> 
> In time, all the useful information about the 'bot should probably be in the repo's wiki. 

Agree.
> Perhaps that would be a good landing page.

I WAG there will likely be a reasonable landing place in a wiki for each major revision (UKMaRSBot-1, UKMaRSBot-2, etc) of the robot.
> 
> The URL needs to be short really. 

Agree. It needs to be printed as big as practical.

> I am keen that even relatively poor photos will have a legible URL to guide people to further information. Very often one sees pictures of interesting things with no clear way to find out more.

Agree. The ukmars url is infinitely better than nothing. However, I think adding a second URL which is even more direct would be good too.

Use a URL shortener, eg. one of these:
https://www.intelligenteconomist.com/url-shortener-services/
several allow the link to be customised, so it needn't be gobbledygook. 

> 
> Both locations will be covered up in a race-ready 'bot. 

I assumed that.

> I will probably place another underneath long with the society logo. 

That's as good as I was hoping for.
> our own sensor boards should probably carry a replacement for the name and link if they cover up the one on the main board.

Ideally yes, though it might be very hard to achieve.

As a side question, is the battery holder a 3D printed part? 
If it is, maybe it could carry the URL, embossed or raised, and filled-in with a sharpie. Then a picture of the 'bot without the battery would be useful.