ukmars/ukmarsbot #8: Add 'ON' to PCB silkscreen next to DIP switch in appropriate position

gbulmer, 2019-08-02:
Revision 1?

I assume the DIP switch used has a clear 'ON' on the body. However, the paint wore off on a (cheap) type I've had. So an extra legend on the PCB might help.

There is an argument that adding a legend to the PCB would be confusing if the DIP switch is soldered in rotated. However, it is likely that documentation for UKMaRSBot will show DIP switch positions, so encouraging the constructor to orientate the switch to match is 'a good thing'.

This would impact the PCB CAD.

micromouseonline, 2019-08-02:
The 'ON' position was on layer tDocu. I will move it to tPlace so that is shows on the silkscreen it will be under the part but, by then, it will not matter. 

Documentation

gbulmer, 2019-08-02:
Okay, that improves the chances of documentation and robot matching.

I still think an 'OFF' visible on the PCB, but if the DIP switch legends are reasonable quality, it's not essential.

closed 2019-08-02 by gbulmer