ukmars/ukmarsbot #9: Add Copyright and License to PCBs silkscreen

gbulmer, 2019-08-02:
Revision 2.

The PCBs are major tangible entities. They should carry clear Copyright and License statements. Then anyone seeing the PCBs or pictures of the PCBs are clear on Copyright and License, and hence whether they are appropriate for their potential use.

This would impact the PCBs CAD.

micromouseonline, 2019-08-02:
If we were to adopt an OSWH designation, that should be sufficient without cluttering up the board?

gbulmer, 2019-08-02:
The OSHW logo isn't a License, it just communicates that the License is compatible with OSHW principles.
I think License and Copyright should be clear and specific.

Right now, I believe the artefacts are on an MIT license because that's what the Github repo says, and the Copyright is the authors.

We don't think we need to rush this.