ukmars/ukmarsbot #16: There appears to be a stray via on Line Sensor PCB

gbulmer, 2019-08-03:
Priority: Low. I don't think it will affect the electronics.
Urgency: Revision 2?
Artefacts impacted: line_sensor_1p1_brd.pdf and its underlying PCB ECAD.

See the attached pdf with an annotation around the via.
[line_sensor_1p1_brd-gb20190803.pdf](https://github.com/ukmars/ukmarsbot/files/3463979/line_sensor_1p1_brd-gb20190803.pdf)

The via is on the lowest left-hand track, between LED3 and the transistor.

IanSuper7, 2019-08-03:
This is J1 on the schematic that provides the option of linking to ground and omitting the LED driver transistor. Using the link instead of the transistor puts the emitters into "always on" and no need for programatic control of the emitters. Builders for some school clubs have asked for this capability although normal use is expected to be pulsing the emitters allowing readings to be taken lit and unlit, reducing power consumption and radiation from the robot.

gbulmer, 2019-08-03:
Okay. I understand.

Please add a label, and maybe a 'box' around the two pins, on the silkscreen to make it clear.

I don't mind what you choose to do. it needs to be clear and tie to the schematic and documentation.

One convention is to use a resistor footprint, put a dotted box around it, and add some text to identify it. I've seen 'NO_POP' (No populate) or 'DNF' (Do Not Fit) as the value.

I'm happy for you to close this.

gbulmer, 2019-08-04:
Looking more closely, if J1/J2 is connected, then as well as T5, R4 and R11 may be missed off too.
 
So it would be helpful to add a 'box' around all the parts involved on both the schematic and PCB.

IanSuper7, 2019-08-04:
v1.2 of the board will expand the narrative around parts that may be optionally omitted and the board silkscreen shows the position of the optional link with description.

micromouseonline, 2019-08-07:
I believe this is now satisfied

gbulmer, 2019-08-07:
Agree. I'll close.

closed 2019-08-07 by gbulmer