ukmars/ukmarsbot #17: Battery holder missing from BoM

gbulmer, 2019-08-03:
**Project phase:** Revision 1
**Artefacts affected:** docs/UKMARSbotBOM.xlsx

There is a PP3 batter holder shown in the renders (eg. renders/ukmarsbot-v45-1.png).
However, AFAICT it isn't mentioned in the BoM (docs/UKMARSbotBOM.xlsx)

It might have a part number and supplier, or be a custom 3D printed part. In either case, it should be identified in the BoM.

micromouseonline, 2019-08-03:
The BOM is not a working, up-to-date document. It was generated before the PCB existed and independently of it.

We need to sort out a whole bunch of issues around that.

Format, presentation, suppliers, how to handle alternatives, what to include. It is a whole big thing all on its own.

For my part, I think there should be a simple component level BOM just for the mainboard without supplier information or part numbers. 

A (separate ?) text document could list suppliers and the range of parts for which that supplier might be useful. Maybe even a matrix of part or part type and suggested suppliers.

I don't know.

Anyway, as for the battery holder, it does not exist and appears on my 3D renders as a physical placeholder, or suggestion, to show where the battery would go without obscuring the view overmuch. Perhaps a transparent PP3 battery solid would be better.

Those 3D renders are, in any case illustrative of a possible prototype configuration, not a 'bot built on the intended PCB.

If you are interested in taking the Eagle files, and generating a better 3D render that we could use, that would be a big help. 

—
Peter Harrison
peter.harrison@helicron.net

gbulmer, 2019-08-04:
> The BOM is not a working, up-to-date document. It was generated before the PCB existed and independently of it. 

Okay. I can understand that.

However it is useful even in its current state.
So a placeholder in the BoM would be enough for this stage.

> We need to sort out a whole bunch of issues around that. Format, presentation, suppliers, how to handle alternatives, what to include. It is a whole big thing all on its own.

I appreciate a general purpose BoM is more complex than the BoM for a single specific build.

AFAICT the goal at the end of this phase is to have some students at one school build the prototype.
I imagine someone will create a BoM for that, and maybe order the parts for the activity.
Hence a 'general purpose, publication ready' BoM doesn't appear to be needed yet.  

A list with a row for each part, generated directly from the Eagle ECAD using the 'bom.ulp' would be okay for now.

For full disclosure. I looked at the PCB first, and saw the battery holder outline. I couldn't figure out  what it was as it seemed to be too narrow. It wasn't clear to me how it was attached either. So I only looked at the render for more clues. 

The render is nice to have. However, to me, less important than the PCB CAD and Eagle BoM. 

If the BoM had said 'Battery Holder' and 'TBD' I would not have raised an issue about it.

gbulmer, 2019-08-04:
For my purposes, a pair of slots through the PCB for Reusable Cable Ties to hold the battery would be good enough. If you could find space, I'd like this option anyway.

I prefer off-the-shelf parts to custom made, eg. 3D printed, parts whenever feasible.

micromouseonline, 2019-08-04:
There is no battery holder. 

Even on the PCB, there is simply a rectangle denoting the battery position.

Builders have at least three options:

- self-adhesive velcro strips.
- poster putty
- solder bent wire loops (paperclips are good) into the holes provided and use an elastic band.

The motor connectors and the placement of the other connectors around the battery space is designed to help locate and support the battery in its intended position. A perfboard prototype illustrates the use of loops and elastic.

—
Peter Harrison
peter.harrison@helicron.net

micromouseonline, 2019-08-04:
picture of [battery clip idea] (https://github.com/ukmars/ukmarsbot/blob/master/renders/ukmarsbot-battery-retainer-suggestion.png) is in renders folder.

[image]

gbulmer, 2019-08-04:
That photograph is extremely helpful. I get it. It removes my immediate concern.

**Phase:** Conclusion of Revision 2

Add a 'README.md' file in the ukmarsbot/renders directory. 

Well-chosen names for the schematics and PCBs mean they likely won't need any explanation. However it would help to explain the meaning or relevance of the 3D renders, photographs. That especially applies to the battery clip which looks like a 'physical part' from the 3D renders. It was those 3D renders that continued my confusion and so I raised this issue.

Also, by the conclusion of Revision 2, add some indication on the UKMARSBot PCB that the rectangle is the battery outline. Maybe make it dotted and have the text "Outline of PP3 Battery"?

micromouseonline, 2019-08-04:
Pretty much all the contents of the repo are currently items for illustration for us and do not necessarily form part of the documentation. Almost none of which has been written.

In particular, a builder's guide will be needed and that will go in the wiki where I would expect photos to be used where appropriate.

The 3D renders are only there because it is easier than trying to describe something which has no physical reality.

At some point, I may make them more functional and descriptive of what will actually be built but there is currently no automatic link between them and the other files so it all has to be done by hand.

Thinking of a builder's guide, would you like us to send you a board so that you can be a test builder and contribute to that part of the project?

—
Peter Harrison
peter.harrison@helicron.net

gbulmer, 2019-08-04:
Here's a stab at a README.md for this directory.

[gb-renders-README.txt](https://github.com/ukmars/ukmarsbot/files/3465025/gb-renders-README.txt)

micromouseonline, 2019-08-04:
Thank you. I have added that.

gbulmer, 2019-08-04:
That resolves this issue for me.

closed 2019-08-04 by gbulmer