ukmars/ukmarsbot #18: Align Part Name conventions between UKMARSBot and Line Sensor

gbulmer, 2019-08-04:
**Phase:** Before PCB for school trials are manufactured
**Artefacts impacted:** either UKMARSBot ECAD, or Line Sensor ECAD, or both

UKMARSBot uses 'LED' as the Part Name prefix for LEDs, and 'D' as the part name prefix for other types of diode.

Line Sensor uses 'LED' and 'D' as the Part Name prefix for LEDs, and has no other types of diode.

**Issue:** Part Naming should be consistent because the same people may be building both.

_Comment_
I prefer 'LED' for all form of LEDs, visible or infrared, and 'D' for other types of diode (eg. junction or Schottky). However, _consistency_ across the ECAD is important to help the school children who will be assembling the PCBs.

In my experience many school children 'know' what an LED looks like, but aren't familiar with physical diodes (some might know its theory). Hence I recommend using 'LED' for LEDs and 'D' for diodes.

micromouseonline, 2019-08-07:
Please re-check for consistency of naming. I believe this to be greatly improved.

gbulmer, 2019-08-07:
Agreed. That is clearer. 
I might nit pick about 'Tn' (phototransistor) vs Q1 (bipolar transistor). However, I am okay with that.

So, I will close this.

closed 2019-08-07 by gbulmer