ukmars/ukmarsbot #20: Add diode-protected power (V_BATT) pin to UKMARSBot sensor header

gbulmer, 2019-08-05:
**Phase:** Revision 2
**Artefacts impacted:** ukmarsbot-brd-20190804A.pdf and underlying UKMARSBot PCB ECAD

**Suggestion:** The sensor board could half its power needs by using LEDs in series running at 7V+ (V_BATT). It would also help sensor boards which would benefit from their own regulators.
 
Further, the sensor board currently draws about (SWAG) 130mA from the Arduino regulator. That's about 4x, or more, the ATmega's requirements. Taking demand off that regulator would help keep it cooler and maybe reduce electrical noise for analogue signals. It would also slightly increase battery life.

Using V-BATT for emitters may be helped because of the BATTERY_VOLTS monitor on  A7.

Using V-BATT on a sensor board would be an option, and not a necessity. Hence there is no built in assumption that the Line sensor, or Wall sensor, needs to be re-engineered to use it.

micromouseonline, 2019-08-05:
Driving the emitters direct from the battery without some kind of constant current source is risky because the battery voltage will fluctuate as the robot runs. Not only will the voltage decrease with state of charge but high power motors and high accelerations will make it sag significantly over shorter intervals. 

It is not clear how to perform reliable compensation of sensor readings for fluctuations in supply voltage so a regulated emitter supply is used.

The regulator on the (genuine) Arduino Nano is an LM1117 and is rated as good for 800mA. Anticipated emitter currents are much lower and are unlikely to exceed 200mA for short periods.  Prolonged currents in excess of, say, 150mA may well cause heating. It remains to be seen if that is problematic but pulsing of the emitters will ensure that the average current is kept modest.

The LM1117 is a good regulator and is capable of better than 50dB ripple even at its maximum 800mA load.

gbulmer, 2019-08-05:
> Driving the emitters direct from the battery without some kind of constant current source is risky 

Both constant current source _or_ on-sensor-board power supply seem reasonable.

> because the battery voltage will fluctuate as the robot runs. Not only will the voltage decrease with state of charge but high power motors and high accelerations will make it sag significantly over shorter intervals.

So maybe have two diode protected power sources, one as now V_BATT to the MCU and sensors, and the second diode protected supply to the motor drive?

micromouseonline, 2022-08-16:
Keeping it simple for now. It all works well enough.

closed 2022-08-16 by micromouseonline