ukmars/ukmarsbot #21: Typos in

gbulmer, 2019-08-08:
**Phase:** Revision 1
**Artefacts impacted:** renders/wall-sensor/README.md

**Issue:** "mainboard outline ad so will not interfere with motion insde the maze" has two typos:
typo "ad" -> "and"
type "insde" -> "inside"

ceeaspb, 2019-08-09:
I had a look at typos also with `aspell` :

**./renders/mainboard/README.md**
Pololu micrometal gearmotors or `equivalen`
The board can `accomodate` wheels
for anti-tipping  skids `usingthe` 3.2mm

**./docs/readme.md**
Various bits of support `documentstsion`

**./docs/encoder-software.md**
and the distance `traveled`
Changes from high to low are `caled`
update the `gloal` counter
The `pololu` `dta` indicates (capital P)
performance `tradeoffs` (suggests space separated or hyphen)

**./docs/analogue-read.md**
and 13 cycles `therafter`
to an ADC and `thed` read in quick `successsion`
Each `chanel` in turn
the `ammount` of noise
`Tht` means it is possible

**./docs/speed-performance.md**
`Practicaly`, the choices
generate the `highst`
the magnetic `ecoder`

**./docs/arduino-pinouts.md**
`Tha` arduino nano (possibly capitalise)
30 KB + 2k `booltolader`
It is `probaby` best
for use `al long`
user input is `desireable`

**./README.md**
to `utilize` these (depends whether US or GB)

**./CONTRIBUTING.md**
seeking specific `focussed` feedback
these `focussed` feedback periods

I ignored a few around capitalisation of ATMega and similar.

micromouseonline, 2019-10-17:
You guys know you can clone the repos, make changes and submit a pull request - right?

ceeaspb, 2019-10-23:
yes I can probably pick this up

closed 2022-08-16 by micromouseonline