ukmars/ukmarsbot #28: SFH4550 not suitable for beginner projects

IanSuper7, 2019-09-11:
To avoid potential optical hazard through incorrect implementation the SFH4550 should not be used for beginner projects. The SFH4550 works well but is a narrow beam IR emitter so has to be used with care with a small duty cycle. It also has 3 variants with different emission levels that could confuse builders on the emission from their robot. For a starter robot, devices requiring less attention to emission safety should be used.

TLCR5800 paired with BPW85C and OSHR3131A-NO (Rapid part no 55-0066) paired with BPW85 detectors (Farnell part 104-5380) are being tested.

The production reference build will include different emitters.

Whatever devices are chosen, builders should assure themselves of the safe emission level of their robot.

ceeaspb, 2019-09-15:
Hi Ian, is this for line sensors, wall sensors or both? thanks

micromouseonline, 2019-09-15:
Both really.

They will be fine if you understand the issues around their use but if you have younger or inexperienced builders, the visible light sensors will be less risky as well as easier to comprehend.

—
Peter Harrison
peter.harrison@helicron.net