ukmars/ukmarsbot #33: Resistor configuration

RomarioKavin1, 2023-05-16:
What should be the resistor configuration for r1-r8 if TLCR5800 is used along with BPW96 instead of SFH4550 and TEFT4300 emitter and transistors in the wall sensor advanced board?

micromouseonline, 2023-05-16:
These resisistors are all adjust-on-test. You will need to try some values and see how you get on. The aim is to make the emitters as bright as possible without overloading them. Take care to ensure they are never left running continuously. 

For the detectors, you want the lowest value load resistor that gives a good range of results. With the board placed so that the detector gives the largest possible result, that value should be no more than about 80% of full scale. In that way you have some headroom for when/if the detectors are operating in a very bright environment.

RomarioKavin1, 2023-05-16:
Can you give me the resistance values used for the Dorothy Bot?

micromouseonline, 2023-05-16:
Sorry - I gave that one away and did not write down the values.

RomarioKavin1, 2023-05-21:
Hey I just Built a new UKMARS bot with the advanced wall follower module,is there any full fledged code to run it

micromouseonline, 2023-05-21:
There is a version of the code for four wall sensors here:
https://github.com/micromouseonline/ukmarsbot-clara

RomarioKavin1, 2023-05-21:
The link seems to be broken or the repo might be private

micromouseonline, 2023-05-21:
My mistake. Try this one:
https://github.com/micromouseonline/mazerunner-core

This is not proven code though versions of it have run successfully in robots in a number of recent contests.

It is subject to sudden, large changes so take care if you pull a version later.

RomarioKavin1, 2023-05-21:
I've tried this code but it doesn't seem to be working on my advanced wall follower module

micromouseonline, 2023-05-21:
It is not meant to be plug-and play. All the code is meant to be for guidance rather than a working solution.

RomarioKavin1, 2023-06-06:
I still can't seem to figure out a code solution. Is there anyway I could possibly get an alternative code? The main bot is not the issue the advanced wall follower seems to be causing problems

micromouseonline, 2023-06-07:
Perhaps you could describe in more detail exactly what the issue is?

RomarioKavin1, 2023-06-08:
No values are being read from the receiver its a constant zero

micromouseonline, 2023-06-08:
Do you have access to an oscilloscope?
If not, do you have a multimeter?

If you read zero at all times then it is likely that there is some other fault. Maybe the detectors are connected the wrong way round? With a multimeter, you should see some voltage on the detector load resistor. That voltage will change when you shine a bright light onto the detector.

RahulRST, 2023-06-22:
should the detector's negative terminal be connected to the board's positive terminal? because my detector is reading some values when light is shined at it but the arduino keeps reading only zero value and another issue is that my emmiter is always on but as far as I know it shouldn't do that and it should only be on when it is required by the code.

micromouseonline, 2023-06-22:
We are still talking about the wall-sensor-advanced design right?
This one:
https://github.com/ukmars/ukmarsbot/tree/master/hardware/wall-sensors/wall-sensor-advanced

The detector is  phototransistor. It must have the COLECTOR connected to supply positive and EMITTER connected to the load resistor. The voltage measured across the load resistor should vary with the amount of light shining on the phototransistor. Try this with the sensor board disconnected from the robot (use a jumper wire to connect only the power leads) and with it plugged in. The behaviour should be the same. If it is not, thene you may have a soldering problem on the mainboard.

While you have the sensor board disconnected, check the emitters. They should all be off. If not, you have a soldering error on the sensor board. If the emitters are always on when the sensor board is connected to the mainboard, try with the Arduino disconnected. If the emitters are still on, the problem is with the mainboard wiring. If it is only happening when the Arduino is plugged in, it is likely that you have a software problem.

RomarioKavin1, 2023-06-22:
What must be the value of the load resistor for the bpw96B photo transistor?Is there a code to just test out the wall follower module? The phototransistor when tested separately reduces in resistance when light is shined on it ,but there are no values when soldered to the pcb and tested on code. It'd be very helpful if we could get the Resistance values used in the Dorothy bot.

micromouseonline, 2023-06-22:
The schemtatic shows values of 1k8 for the detector load reistors. Start with those values. Change them as needed to get an appropriate response. Another board I have uses 1k0 load resisotrs. Every build is likely to be different a we have no control over the exact devices that you choose.

It sounds like you have the phototransistors soldered in the wrong way round.

micromouseonline, 2023-06-22:
There is sample code for nearly everything:
https://github.com/ukmars/ukmarsbot-examples
https://github.com/ukmars/ukmarsbot-mazerunner

RomarioKavin1, 2023-06-27:
Hey,We got the wall sensor board to work .but now in our main board the function switch resistors are acting a bit weird. The values of the resistors increase slowly till they reach their actual value. Example: the resistor r1's value starts at 6.5k ohms slowly climbs to 10k ohms . This leads to in consistent resistance values for the board while using the dipswitch. We've tried changing the resistors multiple times,We even started with a fresh pcb but ran into the same issue the resistors show accurate resistance values once removed from the board but as soon as they are connected in series in the board this problem occurs .Any idea what the problem might be?

micromouseonline, 2023-06-27:
What did you do to make the sensor board work?

RomarioKavin1, 2023-06-27:
We recreated the schematics on a breadboard and adjusted the emitters and collector pins properly then tweaked the resistances a bit and it worked. Any suggestions for the function switch situation?

micromouseonline, 2023-06-27:
But did that result in the actual sensorboard working after you adjusted the component values.

There have been many UKMARSBOT mainboards built without these issues. The board layout and resistor values for the switches are all proven correct. I suspect either that the boards are not being soldered correctly or the microcontroller is somehow faulty. As far as I am aware, you are the only builder that has experienced these problems.

RomarioKavin1, 2023-06-28:
Were the values tested in the mainboard v1.3a

micromouseonline, 2023-06-28:
yes

tgmaxx, 2023-07-08:
> These resisistors are all adjust-on-test. You will need to try some values and see how you get on. The aim is to make the emitters as bright as possible without overloading them. Take care to ensure [...]

I have never seen and/or built a project from plans where the author says you just need to "try some values." If that's how the world of hobbyist electronics worked, there would be no hobbyist market.

I started this endeavor after seeing the Vertasium video on Youtube about the Fastest Maze-Solving Competition On Earth. It looked like the github documentation was OK and I could go ahead and get boards and parts. I was excited after seeing the video. I watched it many times, and watched all the other videos regarding the subject. But while waiting for boards and parts, and the more I got into it, I realized the UKMARSBOT was going to be a big headache. I am a veteran Arduino hobbyist, having spent over 10 years with almost every board they designed.

I also built the Basic Wall Follower and used TLCR5800 with BPW96 instead of SFH4550 and TEFT4300. I don't see why it is not possible to provide resistor values for those parts. Tolerances for the parts is not so wild that providing approximate values can not be done. This is not rocket science.

I would assume there are many others who watched that Vertasium video who are also attempting to build the UKMARSBOT. For the sake of the project, I hope the BOM's, schematics and other documentation can be improved.