ukmars/ukmarsbot #34: Is TEFT4300 and SFH4550 a good match?

MrPec, 2024-01-09:
Hello,

I just built the robot v1.3a along with the wall-sensor-advanced. I checked the datasheet after I got very low readings - and I saw that the TEFT4300 works in the spectrum of 875 to 1000nm, wheras the SFH 4550 has its peak at 860nm. Do I misinterpret something?

Thanks!

Alex

micromouseonline, 2024-08-02:
Sorry - I missed this issue.
They are not a great match. I need to update the details.
A better match would be the SFH4545 or SFH4511 for the transmitter or a BPW85C for the detector.

diberben, 2024-09-16:
Can confirm the mismatch in spectral sensitivity between SFH4550 and TEFT4300. Choice of VSLY5940 for the emitters would be a better match.

micromouseonline, 2024-09-16:
Thank you both.

New updates pushed to repository

closed 2024-09-16 by micromouseonline