ukmars/ukmarsbot #35: Question regarding correct part numbers and values

binaryexistence, 2024-07-31:
Hi there,
I am trying to assemble the PCB of the main board 1.3a. However, in the BOM here on GitHub are other part numbers and resistor/capacitor values mentioned than on the website:
e.g. C2, C3 are 470 µF 10 V on the website, 300 µF 16 V on GitHub,
D1/D3 are 1N5817/1N4001, here 1N5401/1N4148,
IC1 is 74LS86N, here 74HC86,
resistor values are also mixed.

Can you tell me please, which BOM I should stick to?
Thanks in advance!

micromouseonline, 2024-08-01:
Ah. thank you for pointing this out. these bits of content are not all generated by one person.  
The build instructions are a bit out of date. Or they were. I have now edited the 
[Tools and Materials](https://ukmars.org/projects/ukmarsbot/ukmarsbot-build-guide/tools-and-materials/) page to make it easier to read and (hopefully) consistent wit the BOM on github.

That does not mean it is correct so do please point out any further issues.

binaryexistence, 2024-08-02:
Hi! Wow, thank you very much! Now it really looks cleaned up. I will try out with the listed components.
Kind regards

micromouseonline, 2024-08-02:
Raise a new issue if there are any problems.

closed 2024-08-02 by micromouseonline